crownstone/bluenet #13: getting empty CMAKE_OBJCOPY_PAYLOAD

dhruvagarwal, 2015-02-14:
Hey i forked your repo and striped it out to minimal features that i required , now while making the file 
The make goes 100% but in one of the last steps
https://github.com/mrquincle/bluenet/blob/master/CMakeLists.txt#L198

I get an empty CMAKE_OBJCOPY_PAYLOAD 

Here's my github gist of CMakeLists,CMakeBuild.config and main.cpp ( which in your case is cs_main_crownstone.cpp) . https://gist.github.com/dhruvagarwal/47ef318f872dbb091740

I am using SDK from here : https://github.com/mrquincle/nrf51-dfu-bootloader-for-gcc-compiler/tree/master/nrf51822_v6.0.0%20-%20GCC_Bootloader 

I have also commented out a few executables which I was not able to "make" successfully (for eg: drivers/*.cpp ) , as you can see here https://gist.github.com/dhruvagarwal/47ef318f872dbb091740#file-cmakelists-txt-L274 

Please suggest any suggestions.. i need to resolve it ASAP.

mrquincle, 2015-02-14:
Are you sure objcopy and objdump exists on your system? Should be something like `/opt/gcc-arm-none-eabi-4_8-2014q3/bin/arm-none-eabi-objcopy`.

I don't see anything of your compile output in your gists, so it is hard to see what goes wrong.

That would you call an SDK is not an SDK, it is a bootloader. You'll have to choose the `s110` branch if you need the bootloader for the `s110`. This will be merged into the main branch in our next internal sprint.

dhruvagarwal, 2015-02-14:
sorry my bad, i have used the s110 branch only, you are right, the objcopy and objdump are missing from my system. Can you please suggest what fix should i do ?

closed 2015-02-15 by dhruvagarwal