crownstone/bluenet #14: getting assembler messages/errors.. need solution

dhruvagarwal, 2015-02-17:
Hey, this time i am making a fresh build of your repo as it is ( i have only changed the path to bootloader ,etc. etc.) . I have declared a few variables in the CMakeLists.txt ( some were not inherited from Build.config, i don't know why ) , here's a gist.

I am attaching a screenshot of the error that i got while running "make". ( i ran "make" on this dir before as well ,therefore it's only 3% ).
[image]

I can update the gist with other files as well if you say.
Thanks in advance .

mrquincle, 2015-02-17:
Hi Druv,

Can you type:

    VERBOSE=1 make

Thanks!

dhruvagarwal, 2015-02-17:
Hi Anne,

I have updated the gist with VERBOSE=1 make.

Thanks

mrquincle, 2015-02-17:
Hi Druv,

I don't know why you want to do it with your own scripts, but if you are gonna reinvent the wheel, I can't be of much help of course. You need many more flags when you are cross-compiling:

```
/opt/compiler/gcc-arm-none-eabi-4_8-2014q3/bin/arm-none-eabi-g++   -DAPPLICATION_LENGTH=0x2a000 -DAPPLICATION_START_ADDRESS=0x00016000 -DBLUETOOTH_NAME=\"vlier\" -DBOARD=5 -DBOARD_NRF6310 -DCOMPILATION_TIME=\"2015-02-17\" -DGENERAL_SERVICE=1 -DHARDWARE_VERSION=0x001D -DINDOOR_SERVICE=0 -DMESHING=1 -DNORDIC_SDK_VERSION=6 -DNRF51822_DIR=/opt/nrf51_sdk/v6/nrf51822 -DPOWER_SERVICE=1 -DSERIAL_VERBOSITY=1 -DSOFTDEVICE=s110_nrf51822_7.0.0 -DSOFTDEVICE_NO_SEPARATE_UICR_SECTION=1 -DSOFTDEVICE_SERIES=110 -std=c++11 -fno-exceptions -fdelete-dead-exceptions -fno-unwind-tables -fno-non-call-exceptions -g3 -Wall -Os -fomit-frame-pointer   -mcpu=cortex-m0 -mthumb -ffunction-sections -fdata-sections  -O3 -DNDEBUG -I/hd/home/anne/myworkspace/ble/bluenet/include -I/hd/home/anne/myworkspace/ble/bluenet/include/third -I/hd/home/anne/myworkspace/ble/bluenet/include/third/nrf -I/hd/home/anne/myworkspace/ble/bluenet/include/third/protocol -I/opt/nrf51_sdk/v6/nrf51822/Include -I/opt/nrf51_sdk/v6/nrf51822/Include/app_common -I/opt/nrf51_sdk/v6/nrf51822/Include/ble -I/opt/nrf51_sdk/v6/nrf51822/Include/ble/ble_services -I/opt/nrf51_sdk/v6/nrf51822/Include/gcc -I/opt/nrf51_sdk/v6/nrf51822/Include/sd_common -I/opt/softdevices/s110_nrf51822_7.0.0_API/include    -MMD -DNRF51 -DEPD_ENABLE_EXTRA_RAM -DNRF51_USE_SOFTDEVICE=1 -DUSE_RENDER_CONTEXT -DSYSCALLS -DUSING_FUNC -o CMakeFiles/crownstone.dir/src/util/cs_Error.cpp.obj -c /hd/home/anne/myworkspace/ble/bluenet/src/util/cs_Error.cpp
```

This is the set of flags that are set in `arm.toolchain.cmake`.

Good luck

dhruvagarwal, 2015-02-18:
Thanks Anne .. that worked :+1:

closed 2015-02-18 by dhruvagarwal

mrquincle, 2015-02-18:
You're welcome. ;-)