crownstone/bluenet #15: Linking error

dhruvagarwal, 2015-02-18:
I am getting a
collect2: error: ld returned 1 exit status 

This is a gist of my make VERBOSE=1 file and my cmakelists.txt : https://gist.github.com/dhruvagarwal/968f5af6d7ff791a49ef
Please suggest if any flag or something else is missing .

PS: this is hopefully the last problem i am reporting and disturbing you while building this amazing project.

Also, https://github.com/mrquincle/bluenet/blob/master/src/cs_main_crownstone.cpp#L24 will this work ?

mrquincle, 2015-02-18:
The macro `IBEACON` might work indeed. If not it was just a small thing w.r.t. finding a scrambled function if I remember.

Your linker script is also not using the flags from this project. So also those you will need, especially `  -lstdc++_s` and ` -lc_s` . That's why you get errors about `_exit` etc.

```
/opt/compiler/gcc-arm-none-eabi-4_8-2014q3/bin/arm-none-eabi-g++   -std=c++11 -fno-exceptions 
-fdelete-dead-exceptions -fno-unwind-tables -fno-non-call-exceptions -g3 -Wall -Os 
-fomit-frame-pointer   -mcpu=cortex-m0 -mthumb -ffunction-sections -fdata-sections  -O3 
-DNDEBUG  -L/home/anne/myworkspace/ble/bluenet/conf -TnRF51822-softdevice.ld -Wl,--gc-sections 
-Wl,-z,nocopyreloc -Wl,-Map,prog.map -Wl,--wrap,__aeabi_unwind_cpp_pr0 
ALL_OBJECT_FILES 
-o crownstone  -lstdc++_s   -lc_s 
```

Good luck

dhruvagarwal, 2015-02-18:
I am getting these errors on including these flags .
     /usr/bin/../lib/gcc/arm-none-eabi/4.9.3/../../../../arm-none-eabi/bin/ld: cannot find -lstdc++_s
     /usr/bin/../lib/gcc/arm-none-eabi/4.9.3/../../../../arm-none-eabi/bin/ld: cannot find -lc_s

mrquincle, 2015-02-18:
Sure, those are libraries. You'll have to adjust your `LD_LIBRARY_PATH` or indicate via `-L` where the linker can find them.

dhruvagarwal, 2015-02-18:
I don't get exactly where should i add that ?

mrquincle, 2015-02-18:
Just run `locate stdc++_s`. 

```
/opt/gcc-arm-none-eabi-4_8-2014q3/arm-none-eabi/lib/libstdc++_s.a
/opt/gcc-arm-none-eabi-4_8-2014q3/arm-none-eabi/lib/armv6-m/libstdc++_s.a
/opt/gcc-arm-none-eabi-4_8-2014q3/arm-none-eabi/lib/armv7-ar/thumb/libstdc++_s.a
/opt/gcc-arm-none-eabi-4_8-2014q3/arm-none-eabi/lib/armv7-ar/thumb/fpu/libstdc++_s.a
/opt/gcc-arm-none-eabi-4_8-2014q3/arm-none-eabi/lib/armv7-ar/thumb/softfp/libstdc++_s.a
/opt/gcc-arm-none-eabi-4_8-2014q3/arm-none-eabi/lib/armv7-m/libstdc++_s.a
/opt/gcc-arm-none-eabi-4_8-2014q3/arm-none-eabi/lib/armv7e-m/libstdc++_s.a
/opt/gcc-arm-none-eabi-4_8-2014q3/arm-none-eabi/lib/armv7e-m/fpu/libstdc++_s.a
/opt/gcc-arm-none-eabi-4_8-2014q3/arm-none-eabi/lib/armv7e-m/softfp/libstdc++_s.a
/opt/gcc-arm-none-eabi-4_8-2014q3/arm-none-eabi/lib/fpu/libstdc++_s.a
/opt/gcc-arm-none-eabi-4_8-2014q3/arm-none-eabi/lib/thumb/libstdc++_s.a
```

The entire reason I have build this project is so people do not need to set up all this themselves. Why don't you just use our scripts!? In my opinion you are really wasting your time by doing what I did again, but now in your own Makefiles... What are gonna do if there is a new softdevice coming out? Or if you want to switch to another compiler over a year, such as `clang`? I think it's wiser of you to adjust the list files you want to compile in `CMakeLists.txt` and use the build system as is.

dhruvagarwal, 2015-02-19:
Thanks for the help Anne,

When i ran locate stdc++_s, i got just

	/usr/share/doc/gcc-4.8-base/C++/libstdc++_symbols.txt.amd64

I understand what you are saying (about not wasting the time in this whole already done things) and i appreciate the fact that you have made the system so generic and ready to use anywhere. The point that i am making this all again is just to learn, in the process of fixing the errors i have learnt a lot of things from you and if any other issues i search and resolved, if i was to directly use this , i would have directly used your scripts.

Thanks a lot!

mrquincle, 2015-02-19:
Okay, make sure you run `sudo updatedb` to update your database to find files. But apart from that I think you didn't get the cross-compiler in the sense as recommended in the README file. If you get the file from https://launchpad.net/gcc-arm-embedded, in my case I use the slightly older one from Q3, `gcc-arm-none-eabi-4_8-2014q3-20140805-linux.tar.bz2` you will have these files in that tar-ball. However, if you download the cross-compiler via other means I am not so sure.

dhruvagarwal, 2015-02-20:
I downloaded the file from here only but debian package :( 
I ran updatedb but no good results .
Should i reconfig that ?

dhruvagarwal, 2015-02-21:
Please reply if possible ?

mrquincle, 2015-02-21:
Hi Druv,

I've to say I slowly loosing my patience. Can you not just find a friend who explains you the basics about Linux? In the readme it's stated which tar-ball you have to use, and you see that the indicated libraries are part of it by a simple `tar` command:

```
tar -tf ~/setup/gcc-arm-none-eabi-4_8-2014q3-20140805-linux.tar.bz2 | grep stdc++_s
gcc-arm-none-eabi-4_8-2014q3/arm-none-eabi/lib/armv7-ar/thumb/softfp/libstdc++_s.a
gcc-arm-none-eabi-4_8-2014q3/arm-none-eabi/lib/armv7-ar/thumb/fpu/libstdc++_s.a
gcc-arm-none-eabi-4_8-2014q3/arm-none-eabi/lib/armv7-ar/thumb/libstdc++_s.a
gcc-arm-none-eabi-4_8-2014q3/arm-none-eabi/lib/thumb/libstdc++_s.a
gcc-arm-none-eabi-4_8-2014q3/arm-none-eabi/lib/armv7-m/libstdc++_s.a
gcc-arm-none-eabi-4_8-2014q3/arm-none-eabi/lib/armv6-m/libstdc++_s.a
gcc-arm-none-eabi-4_8-2014q3/arm-none-eabi/lib/fpu/libstdc++_s.a
gcc-arm-none-eabi-4_8-2014q3/arm-none-eabi/lib/libstdc++_s.a
gcc-arm-none-eabi-4_8-2014q3/arm-none-eabi/lib/armv7e-m/softfp/libstdc++_s.a
gcc-arm-none-eabi-4_8-2014q3/arm-none-eabi/lib/armv7e-m/fpu/libstdc++_s.a
gcc-arm-none-eabi-4_8-2014q3/arm-none-eabi/lib/armv7e-m/libstdc++_s.a
```

I think it's wise to try find answers to the problems you encounter on other places on the world wide web. They have almost nothing to do with the code in `bluenet`. You're last question especially is just not going through the `README` file but instead downloading a `.deb` file instead of `tar.gz` file which is recommended and then asking me to suggest a solution why certain files cannot be found in the package you installed yourself. Is it really my task to tell you to inspect `.deb` files with `dpkg` and `tar.gz` files with `tar`, etc?

I really want to be as accommodating as possible, but it becomes really hard to see what the added benefit of this approach is to anyway else accept for being your personal teacher w.r.t all kind of things that are Linux-related. I'm also fine with that, but I'm afraid that will come with a price tag. :-)

Good luck with everything!

closed 2015-02-21 by mrquincle