crownstone/bluenet #16: Where can I find the Log messages

dhruvagarwal, 2015-03-26:
Hey [mrquincle] ( https://github.com/mrquincle ) , first of all I would live to thank you for your beautiful code and scripts that helped me understand and use this project.

I need a small help , if you could tell where can i find the Log messages like this one https://github.com/mrquincle/bluenet/blob/master/src/cs_main_crownstone.cpp#L128

Thanks

mrquincle, 2015-03-30:
The `LOG` macro is defined in the file https://github.com/mrquincle/bluenet/blob/master/src/drivers/cs_Serial.cpp. The basic code is:

```
for(int i = 0; i < len; ++i) {
			NRF51_UART_TXD = (uint8_t)p_buf[i];
			while(NRF51_UART_TXDRDY != 1) {}
			NRF51_UART_TXDRDY = 0;
		}
```

The TX pin depends on the hardware you use. This is a configuration option that you can set in https://github.com/mrquincle/bluenet/blob/master/include/common/cs_Boards.h. You need subsequently to read over UART using some kind of hardware you might have available. I use for example the USB dongle from RFduino for it as explained on https://dobots.nl/2014/03/05/rfduino-without-rfduino-code/, but there are almost an infinite number of such boards available. Just pick one that matches your hardware (3.3V or 5V).

dhruvagarwal, 2015-03-30:
Thanks a lot ! :+1:

closed 2015-03-30 by dhruvagarwal