crownstone/bluenet #27: Advertisement

mrquincle, 2016-09-01:
The `updateAdvertisement` function at https://github.com/crownstone/bluenet/blob/sdk_11/src/ble/cs_Stack.cpp#L634-L652 does not respect the fact that the device might NOT be advertising there.

As a workaround I'm not stopping the program if `ble_advdata_set` returns `NRF_ERROR_INVALID_PARAMETER` (0x07), but it should just not enter that function if it's not advertising.

The other functions are organized like this:

- configureX: populate the structs (and allocate if not done before)
- updateX: perform the sd_ble calls that inform the softdevice of the (updated) configuration (but make sure that a mistake is caught, so check if the device name is present, check if we are actually advertising, check if the softdevice is enabled)
- startX, stopX: perform sd_ble calls for starting/stopping
- isX: retrieve local state (the softdevice cannot know if it is advertising, storing this state is the responsibility of the application: https://devzone.nordicsemi.com/question/80959/check-if-currently-advertising/)

Bug: it shouldn't be possible to raise a softdevice error calling `updateAdvertisement`.

mrquincle, 2016-09-01:
The code:

```
void Nrf51822BluetoothStack::updateAdvertisement() {
//      Why disabled?
//	if (!_advertising)
//		return;

	uint32_t err;
	err = ble_advdata_set(&_advdata, &_scan_resp);

	switch(err) {
	//case NRF_ERROR_INVALID_PARAMETER:
	case 0x07:
		LOGi("Invalid advertisement configuration");
		break;
	default:
	//	BLE_CALL(sd_ble_gap_adv_stop, ());
		BLE_CALL(ble_advdata_set, (&_advdata, &_scan_resp));
	//	BLE_CALL(sd_ble_gap_adv_start, (&adv_params));
	}
}
```

mrquincle, 2016-12-21:
This needs still some love.

vliedel, 2019-09-18:
This should be fixed by the advertising refactor e1df303b51c5d937e3998d20e4fe74ecda98f4da

closed 2019-09-18 by vliedel