crownstone/bluenet #29: Characteristic is still trying to use encryption in setup mode

vliedel, 2016-09-08:
[cs_main_crownstone.cpp         : 153  ] Configure setup mode                                          
[cs_main_crownstone.cpp         : 493  ] Create all services                                           
[cs_DeviceInformationService.cp : 32   ] Device Information Service init                               
[cs_DeviceInformationService.cp : 34   ] Characteristic Hardware Revision added                        
[cs_DeviceInformationService.cp : 37   ] Characteristic Firmware Revision added                        
[cs_DeviceInformationService.cp : 41   ] Characteristic Software Revision added                        
[cs_SetupService.cpp            : 26   ] Setup Service init                                            
[cs_SetupService.cpp            : 33   ] Characteristic Control added                                  
[cs_SetupService.cpp            : 36   ] Characteristic MAC Address added                              
[cs_SetupService.cpp            : 40   ] Characteristic Setup Encryption Key added                     
[cs_SetupService.cpp            : 43   ] Characteristic Configuration added                            
[cs_Stack.cpp                   : 311  ] Set tx power to -40                                           
[cs_main_crownstone.cpp         : 168  ] Enable AES encryption                                         
[cs_main_crownstone.cpp         : 212  ] ---- init services ----                                       
[cs_Characteristic.cpp          : 87   ] Hardware Revision init with buffer[20] with 0x200056c8        
[cs_Characteristic.cpp          : 87   ] Firmware Revision init with buffer[20] with 0x200057f8        
[cs_Characteristic.cpp          : 87   ] Software Revision init with buffer[20] with 0x20005810        
[cs_Characteristic.cpp          : 87   ] Control init with buffer[300] with 0x20004b48                 
[cs_Characteristic.cpp          : 87   ] MAC Address init with buffer[6] with 0x20005419               
[cs_Characteristic.cpp          : 87   ] Setup Encryption Key init with buffer[16] with 0x20005660     
[cs_Characteristic.cpp          : 87   ] Config Control init with buffer[300] with 0x20004b48          
[cs_Characteristic.cpp          : 87   ] Config Read init with buffer[300] with 0x20004b48
[cs_EncryptionHandler.cpp       : 434  ] Can't use this setup key                                      
[cs_Characteristic.cpp          : 228  ] error encrypting data.

AlexDM0, 2016-09-08:
The error is more of a side effect of the mechanisms of the encryption. Here, it is harmless. The setup phase DOES use encryption as it should.

vliedel, 2021-05-06:
No longer the case.

closed 2021-05-06 by vliedel