crownstone/bluenet #31: Samsung S5 "within range" functionality stops working

mrquincle, 2017-01-09:
A Samsung S5 smartphone stops with the functionality that is about getting in and out of range.

- On (immediately)
- Off (after 2 minutes)

There might be several reasons:

- Somehow a certain sleep mode (dozing, app standby) might put the app in such a state that it doesn't perform certain functionality, be it scanning, be it connecting: https://developer.android.com/training/monitoring-device-state/doze-standby.html
- The service that is running in the background stops.
- The service that is running in the background stops and restarts, but it doesn't restart the complete app.
- The Android OS turns off certain BLE functionality if it feels like it.
- There is some kind of BLE error that is triggered rarely and screws up the connection state in the app.
- There is a connection set up with a Crownstone that is not disconnected.

AlexDM0, 2017-01-09:
>    training/monitoring-device-state/doze-standby.html

Does it work only once in the background (turns off but does not turn on
again)? iOS had this issue and that's now fixed in the app.

mrquincle, 2017-01-09:
It rarely responds to anything.

I propose some kind of mechanical construction where a Faraday box is placed on top of a smartphone once every 5 minutes, so in an hour there should be 12 x 2 switches in a connected light and in a working day there should be 24 x 8 events. Then the reliability can be assessed a bit.

vliedel, 2017-01-10:
With the foreground service, I have never seen the service being killed yet. What does happen is that bluetooth scanning spits an error and simply stops working (similar to https://code.google.com/p/android/issues/detail?id=173991), this is often accompanied by connect error 257. People seem to advice to simply turn off and on bluetooth programmatically, but i'm trying to find the error in the hci log first.

mrquincle, 2019-09-17:
This will be changed (and very likely solved) with the next firmware update that will have the Crownstone scan for broadcasting devices rather than the phone scan for broadcasting Crownstones.

closed 2019-09-17 by mrquincle