crownstone/bluenet #32: Dummy Platform

mrquincle, 2017-02-13:
Ideally, we would abstract the hardware in such a way that we can also run the code on x86 where:

- the different reads and writes to registers are printed or visualized
- the normal gcc (or clang) can be used, no cross-compilation flags e.g.
- the CTest unit tests can be used to test functionality on the machine of the developer

The way to do this is to provide a clone of the Nordic repository where the header files are almost the same, but the C files are different. The implementation is at times part of the header files, so at times the header files need to be adjusted as well. This will allow anyone developing for the nRF52 to swap out the Nordic repository for this "dummy platform" repository and build everything with only a few changes to the compiler flags.

AlexDM0, 2017-02-13:
That would indeed be cool! Worth looking into I agree!

mrquincle, 2019-09-17:
Better than a dummy platform, we will move towards a continuous hardware integration setup. This will bit by bit take more of the continuous integration tasks that have the "hardware in the loop" out of our hands.

closed 2019-09-17 by mrquincle