crownstone/bluenet #44: Does not compile due to flexible array member in union

mrquincle, 2017-03-11:
Error speaks for itself:

```
In file included from ~/workspace/bluenet/include/mesh/cs_Mesh.h:18:0,
                 from ~/workspace/bluenet/src/drivers/cs_Storage.cpp:27:
~/workspace/bluenet/include/protocol/cs_MeshMessageTypes.h:262:21: error: flexible array member in union
     uint8_t payload[];
```

mrquincle, 2017-03-11:
The easiest way to do this is to wrap a plain old data object and write accessors that can immediately check also if lengths etc. are correct. If you want to be fancy you can overload serialization operators, but just `get` and `set` are sufficiently.

In this gist I make the same assumption on endianness as in the current code. 

https://gist.github.com/mrquincle/c44b5e6422a407d7a406544ad3b2c80e

I think it's fine to have such an assumption as long as at the location where the messages enter this is addressed. If we do not assume this we will need to bit shift all arrays of uint16_t or larger to be independent on the platform endianness.

mrquincle, 2017-03-11:
If we almost never access anything to write the above is overkill.

Shouldn't there be just uint8_t* in that case for such dynamic arrays in unions?

mrquincle, 2017-03-11:
If we almost never access anything to write the above is overkill.

Sizeof uint8_t* is not zero so it's not the same.

On my system uint8_t[0] works though. Anyone a problem with that?

vliedel, 2017-03-14:
So you mean `uint8_t payload[0];` ? That works for me.

mrquincle, 2017-03-14:
Yes

mrquincle, 2017-04-02:
It compiles, but it ain't pretty.

closed 2017-04-02 by mrquincle