crownstone/bluenet #51: No env.config.template file

shkhamd, 2017-05-31:
I am trying to fresh install the repo in my **$HOME** as instructed in the _/bluenet/blob/master/docs/INSTALL.md_ 
> Now we need to set up the environment variables to keep track of the different folders required to build bluenet

> > cd ~/bluenet_workspace/bluenet
> > cp env.config.template env.config

There is no file called  _env.config.template_ and the copy command fails. 
Where should I look for the config template file? 

In _/scripts/env.sh_ I see that loading from env.config is optional. So will the installation work without the _env.config_ ?

mrquincle, 2017-05-31:
Hi! Good catch. We moved all config files out of the root a while ago to conf/ and conf/cmake. The update hasn't been reflected in the docs/INSTALL.md yet. Thanks for updating us!

shkhamd, 2017-05-31:
hi, yeah, just right after, I found them under _~/bluenet/conf/cmake/_ directory. 
I think the cp commands should be updated. 
Thanks.

closed 2017-05-31 by shkhamd