crownstone/bluenet #61: Matter of style / consistency w.r.t. hardware presence

mrquincle, 2018-12-05:
There is an `IS_CROWNSTONE` macro that is used to enable for example modules like switch, pwm etc. There are also fine grained configuration fields such as `_boardsConfig.flags.hasLed`. It would be best to make it the same. In the future there might be Crownstones without IGBTs for example. Or Crownstones without relays. It's best to use the same manner of configuration. That would be something like this:

* Switch: `_boardsConfig.flags.hasSwitch`
* TemperatureSensor: `_boardsConfig.flags.hasTemperatureSensor`
* VoltageSensor: `_boardsConfig.flags.hasVoltageSensor`
* CurrentSensor: `_boardsConfig.flags.hasCurrentSensor`

There are also software modules that we might enable in a similar manner:

* Watchdog: `_boardsConfig.flags.enableWatchdog`
* EnOcean: `_boardsConfig.flags.enableEnocean`

AlexDM0, 2018-12-05:
suggestion for naming: 

_boardsConfig --> _boardConfig since it is the config for this board, boardsConfig is a large file with all supported boards, this will set a single boardConfig for the hardware that is being used.

Maybe remove the .flags? it does not seem to be adding anything to this setup.  Maybe separate between hardware and software:

_boardConfig.hw.hasSwitch

_boardConfig.modules.useEnocean

mrquincle, 2022-09-05:
This is now how it is implemented.

closed 2022-09-05 by mrquincle