crownstone/bluenet #69: There is a short flash when performing DFU

mrquincle, 2019-06-12:
This brief flash (even with a larger load than the LED we normally use), can be caused by:

- A pin that is set to be pulled up while it shouldn't.
- A pin that is floating during the reset.
- A misconfigured pin (the bootloader or the mesh configures this pin thinking it does something else)

Some more context:

- This is even happening when only the bootloader is present. Hence, it cannot be debug pins of the mesh or something like that.
- When running for a while there seem to be spontaneous reset.
- After running for a while it seems easier to trigger the flash. Normally it happens with a DFU update. However, if running for a while it seems easier to trigger with a ./reset.sh execution.
- The flash seems to be happening before hitting SystemInit in the bootloader. After running the bootloader once, it will take a while until the flash is again visible.
- It is harder to check if resets are happening when there's only a bootloader (`./bluenet.sh -t sdk15 -ebudSHB` written because the reset counter is only written in the application.
- It is not necessary to trigger with a DFU update. After waiting for a while (say 30 seconds), just running debugging (`./bluenet.sh -t sdk15 -dB`) is sufficient to trigger a flash of the light.

Hypothesis: It is a **hardware issue**.

The GPIO pin that goes towards the optocoupler can be floating. If it is floating then it will put 3.3V divided by 2 on both inputs of the optocoupler that depending on the state of the dimming circuit can actually drive the LEDs on the gate driver side up sufficiently so they will be both conducting. This means that the IGBTs can be shortly driven by the gate driver depending on the state of the circuitry. The hypothesis is that when programming for the first time this state is different then when the device is running operationally or running for a while. The dimming circuitry takes some time to deliver sufficient power, but we do not remove the power so this on itself cannot be the reason. However, if it is hardware it is important to wait for the final version. It has a clamping mechanism where at least issues around starting up of the dimming circuitry are resolved. Only when there is a sufficient voltage on the gate driver, it will be possible to turn on the IGBTs. 

Note, that this does not actually solve the floating default. The 3.3V of the input of the optocoupler can be delivered by the chip itself (so then it goes floating as well on a reset and there's no 3.3V / 2 on the pins).

mrquincle, 2019-06-18:
Created a `hardware_info.sh` script to parse the FICR registers.

```
Crownstone has IC chip revision 2. The QFN48 package, with 512 kB FLASH and 64 kB RAM denoted by QFAA-Bx0
(with x any digit, in our case a 0)

Nordic hardware version:	 52832
Board variant:			 AAB0
Board package:			 QFxx - 48-pin QFN
Complete board identifier:	 QFAA-B0
Unique device id:		 EBC106E7167DC349
Bluetooth address:		 C2:CD:96:C0:CB:14:AB:68
Address type:			 Random address
RAM:				 K64 (64 kB RAM)
FLASH:				 K512 (512 kB FLASH)
Temperature slope A0:		 320
Temperature slope A1:		 343
Temperature slope A2:		 35D
Temperature slope A3:		 400
Temperature slope A4:		 452
Temperature slope A5:		 CCFFFFF37B
Temperature y-intercept B0:	 3FCC
Temperature y-intercept B1:	 3F98
Temperature y-intercept B2:	 3F98
Temperature y-intercept B3:	 0012
Temperature y-intercept B4:	 004D
Temperature y-intercept B5:	 3E10
Temperature segment end T0:	 E2
Temperature segment end T1:	 00
Temperature segment end T2:	 14
Temperature segment end T3:	 19
Temperature segment end T4:	 50
```

It's not the latest chip revision, but **revision 1**. The errata can be found here: https://infocenter.nordicsemi.com/topic/errata_nRF52832_Rev1/ERR/nRF52832/Rev1/latest/err_832_new.html. There's no obvious chip hardware issue with this revision.

mrquincle, 2019-06-18:
The default pin behavior is hardware defined: https://devzone.nordicsemi.com/f/nordic-q-a/18421/nrf52832-uart-pin-state-in-reset This means that there is no workaround that keeps output pins low (or high) during a reboot. They will become floating.

mrquincle, 2019-06-19:
The cause for the reset seems to stem from the bootloader:

```
Breakpoint 4, do_reset (p_context=<optimized out>) at /home/anne/workspace/bluenet/source/bootloader/nrf_bootloader.c:127
127	    NVIC_SystemReset();
(gdb) bt
#0  do_reset (p_context=<optimized out>) at /home/anne/workspace/bluenet/source/bootloader/nrf_bootloader.c:127
#1  0x00075cfa in settings_write (p_dst=0x7a501 <delay_machine_code>, p_src=0x7f000 <m_dfu_settings_buffer>, callback=0x732ad <do_reset>, p_dfu_settings_buffer=0x200074e0 <m_health_server+4>) at /home/anne/software/nrf5_sdk/nRF5_SDK_15.3.0_59ac345/components/libraries/bootloader/dfu/nrf_dfu_settings.c:316
#2  0x00075de6 in settings_backup (callback=0x732ad <do_reset>, p_src=0x7f000 <m_dfu_settings_buffer>) at /home/anne/software/nrf5_sdk/nRF5_SDK_15.3.0_59ac345/components/libraries/bootloader/dfu/nrf_dfu_settings.c:378
#3  0x000738ae in timer_fire (p_timer=0x200057bc <_acUpBuffer+2799>) at /home/anne/software/nrf5_sdk/nRF5_SDK_15.3.0_59ac345/components/libraries/bootloader/nrf_bootloader_dfu_timers.c:200
#4  RTC2_IRQHandler () at /home/anne/software/nrf5_sdk/nRF5_SDK_15.3.0_59ac345/components/libraries/bootloader/nrf_bootloader_dfu_timers.c:240
#5  <signal handler called>
#6  0x00015cce in ?? ()
#7  0x00015ccc in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
```

mrquincle, 2019-06-19:
It is possible with the JLink ([pdf](http://supp.iar.com/FilesPublic/UPDINFO/008011/arm/doc/jlinkarm.pdf)) to execute several types of resets. Experiment with different `mon reset x` values in `gdb/gdbreset`:

```
target remote localhost:2331                                                                                            
set remote memory-write-packet-size 1024                                                                                
                                                                                                                        
# See http://supp.iar.com/FilesPublic/UPDINFO/008011/arm/doc/jlinkarm.pdf                                               
                                                                                                                        
# Type 0: Normal. The default, works also when the reset pin is not connected (from JTAG)                               
# This does cause a flash                                                                                               
#monitor reset 0                                                                                                        
                                                                                                                        
# Type 1: Core. Resets the core only, not peripherals                                                                   
# This does NOT cause a flash                                                                                           
monitor reset 1                                                                                                         
                                                                                                                        
# Type 2: ResetPin. Resets the core and the peripherals via SYSRESETREQ & VECTRESET                                     
# This does NOT cause a flash (later on, it actually does, probably because of invalid state after the reset)           
# monitor reset 2                                                                                                       
                                                                                                                        
# Type 3: Connect under Reset. Resets using RESET Pin low when connecting to core                                       
# This does cause a flash (big one)                                                                                     
#monitor reset 3                                                                                                        
                                                                                                                        
# Type 4: Reset core & peripherals, halt after bootloader. Resets core & peripherals, halts CPU after the bootloader    
# This does cause a flash                                                                                               
#monitor reset 4                                                                                                        
                                                                                                                        
# Type 5: Reset core & peripherals, halt before bootloader                                                              
#monitor reset 5                                                                                                        
                                                                                                                        
# Type 8: Reset core and peripherals                                                                                    
# For most devices same as monitor reset 0                                                                              
                                                                                                                        
# Type 6,7, and 9 are for specific devices to disable the watchdog timer immediately after a reset, often by            
# a single device specific instruction.                                                                                 
                                                                                                                        
monitor go                                                                                                              
q
```

Check, if `monitor reset 1` actually does reset the device:

```
rm -f tmp/memory; ./printAppData.sh > /dev/null; ./parse-fds.py tmp/memory
empty page
new page
type=  0 len=  4 data: 18 00 FF FF  
type=  0 len=  4 data: 80 FF FF FF  
type=  0 len=  4 data: 19 00 FF FF  
type=  0 len=  4 data: 80 FF FF FF  
type=  0 len=  4 data: 1A 00 FF FF  
type=129 len=  4 data: 80 FF FF FF  [STATE_SWITCH_STATE]
type=128 len=  4 data: 1B 00 FF FF  [STATE_RESET_COUNTER]
empty page
empty page
```

Just resetting the CPU without the peripherals works like a breeze. :-)

mrquincle, 2019-06-19:
Note that this is told by the JLink manual:

> Only the core is reset via the VECTRESET bit. The peripherals are not affected. After setting the VECTRESET bit, J-Link waits for the S_RESET_ST bit in the Debug Halting Control  and  Status  Register  (DHCSR)  to  first  become  high  and  then  low  afterwards.The   CPU   does   not   start   execution   of   the   program   because   J-Link sets the VC_CORERESET bit before reset, which causes the CPU to halt before execution of the first instruction.Note:In most cases it is not recommended to reset the core only since most target  applications  rely  of  the  reset  state  of  some  peripherals  (PLL,  External  memory interface etc.) and may be confused if they boot up but the peripherals are already configured.

mrquincle, 2019-06-20:
**More detailed hypothesis**

Let's indeed suppose this is a hardware issue. Given a circuitry with a protection measure that keeps the IGBTs off when the power to the gate driver is too low. Then we can have a short flash when performing a reset and having no furthermore flashes with subsequent resets. How? Suppose the floating GPIO pin drives enough current for both optocoupler inputs. The outputs short the power to the gate driver. The IGBTs will be on for a short period. The protection measure kicks in when the power drops under a particular threshold. It will take a while before the power to the gate driver is completely up again. Hence, further resets will have impact again only after waiting a while.

kurkesmurfer, 2019-06-20:
Hi Anne,

This was discussed with Anton and he insists we should be able to measure this occurring with the differential probe. The question is how fast the floating GPIO causes the IGBT power supply to fall. It is capacitive in the sense that it requires loading the 47nF again before power is up. 
The threshold voltage for the gate driver to be in control again is 5V. If the gate clamp is too slow to act (which it is not supposed to) there might be an upward hysteresis.
I’ll check this as soon as I am back.

Kind regards,
Peet

> More detailed hypothesis
>
> Let's indeed suppose this is a hardware issue. Given a circuitry with a protection measure that keeps the IGBTs off when the power to the gate driver is too low. Then we [...]

P.M.A.  van Tooren

Almende/Crownstone
Stationsplein 45, D1.118
3013 AK Rotterdam
Groothandelsgebouw 

mobile: +31654984206

mrquincle, 2019-09-17:
This has been solved in v22 by driving the 3V3 voltage from the chip. Now, if both pins are floating, there will be no voltage divider that puts 3V3/2 on one of the inputs of the opto. Moreover, the current with which the opto is driven is reduced by increasing the R6/R7 from 1kOhm to more than a tenfold of that, decreasing current naturally. This solves this problem.

closed 2019-09-17 by mrquincle