crownstone/bluenet #76: nrfutil not installed

PaulWondel, 2019-10-16:
When following the install instructions, I got an error "make[3]: nrfutil: Command not found" at the step "make build_bootloader_settings".

mrquincle, 2019-10-16:
Then the installation did not succeed. You can see that `nrfutil` should have been installed in `https://github.com/crownstone/bluenet/blob/master/CMakeLists.txt`. Note that the installation requires `sudo`. If you run the installation and you have not entered your password this is a tell that something went wrong during the INSTALL step of `nrfutil`. Run `make clean` and `make` again in the `build` directory and make sure everything gets installed.

PaulWondel, 2019-10-16:
I have done the `make clean` and `make` in the build folder. `nrfutil` is still not installed. In the installation it says:

"[ 67%] Creating directories for 'nrfutil'
[ 69%] Performing download step (download, verify and extract) for 'nrfutil'
-- verifying file...
       file='/home/gmprincep/gitfiles/bluenet/downloads/archive.tar'
-- File already exists and hash match (skip download):
  file='/home/gmprincep/gitfiles/bluenet/downloads/archive.tar'
  MD5='545df3556fc1a70da7a99926fb426eb3'
-- extracting...
     src='/home/gmprincep/gitfiles/bluenet/downloads/archive.tar'
     dst='/home/gmprincep/gitfiles/bluenet/tools/nrfutil'
-- extracting... [tar xf]
-- extracting... [analysis]
-- extracting... [rename]
-- extracting... [clean up]
-- extracting... done
[ 71%] No patch step for 'nrfutil'
[ 73%] No update step for 'nrfutil'
[ 75%] Performing configure step for 'nrfutil'
Skipping configure step.
[ 77%] Performing build step for 'nrfutil'
Skipping build step.
[ 79%] Performing install step for 'nrfutil'
(Reading database ... 377042 files and directories currently installed.)
Preparing to unpack .../nRF-Command-Line-Tools_10_3_0_Linux-amd64.deb ...
Unpacking nrf-command-line-tools (10.3.0) over (10.3.0) ...
Setting up nrf-command-line-tools (10.3.0) ...
Processing triggers for libc-bin (2.27-3ubuntu1) ...
[ 81%] Completed 'nrfutil'
[ 81%] Built target nrfutil
[ 83%] Build micro-eec
micro-ecc not found! Let's pull it from HEAD.
Cloning into 'micro-ecc'..."

The installation finishes installing the "[ 83%] Built target micro_eec" but doesn't show if nrfutil is installed or not.

mrquincle, 2019-10-16:
Use `VERBOSE=1 make` and find out the exact command for installing nrfutil.
Run that command separately and see where it installs the files. Run `sudo
updated && locate nrfutil` to see where it went.

PaulWondel, 2019-10-16:
It looks like this installs nrfjprog, not nrfutil.

mrquincle, 2019-10-16:
Ah! Solved in commit https://github.com/crownstone/bluenet/commit/091d6cd075377e5f9ae66c80d3c24cb0f9628ee7

mrquincle, 2019-10-16:
In latest commit `pip3` is used to be future proof. With this version of `cmake` the python (or pip) dependencies are hard to catch. For now, only messages will be shown. 

The main `CMakeLists.txt` in https://github.com/crownstone/bluenet/commit/01a0afef9b81484895fabd20bab0f3a68aff674b is now defined with `NONE` to remove compiler requirements. You should not need `gcc` or `build-essentials` anymore.

closed 2019-10-16 by mrquincle