crownstone/bluenet #79: A CMake target to create a "factory image"

mrquincle, 2019-12-03:
There should be a CMake target to create a factory image. This takes general firmware, bootloader, and softdevice binaries and enriches it with information to flash a factory image.

- It should take previously compiled binaries and not touch them anymore. 
- The binaries should be enriched with UICR registers for the target hardware (e.g. indicating the hardware type, hardware board, patch version, etc.).
- The bootloader address is written in an UCIR register.
- The binaries should be combined in such a way that it forms one deployable zip file.

This information is read from a `config` file (but a different one from the config files used for building).

vliedel, 2019-12-03:
Should also include MBR address.

Binaries to be included:

- softdevice
- bootloader
- firmware
- bootloader settings

mrquincle, 2019-12-04:
The commit https://github.com/crownstone/bluenet/commit/a098ad2ec5faf4ecddd78be9d7be5449124586e9 takes care of this. The https://github.com/crownstone/bluenet/blob/master/docs/RELEASE_VERSION.md file describes some of the configuration options available. For now a https://github.com/crownstone/bluenet/blob/master/factory-images/builtin-one/CMakeFactoryImage.config file is committed to see all options that can be used. We have to decide where to keep this configuration information (e.g. we might not want to commit a new file for every production run with a different year/week number).

closed 2020-03-05 by mrquincle