crownstone/bluenet #82: Propagate user locations through the mesh

mrquincle, 2019-12-06:
**Propagate time information**
Currently time is propagated through the mesh. This makes sure that after a reboot a Crownstone can get the time. This is done through regularly broadcasting a `CS_MESH_MODEL_TYPE_STATE_TIME` type of message (see [cs_MeshModelPackets.h](https://github.com/crownstone/bluenet/blob/master/source/include/protocol/mesh/cs_MeshModelPackets.h)). If a Crownstone does not have a time it can update its internal time using this information. In contrast, when it has time an authoritative entity (like a smartphone) can update the time using  `CS_MESH_MODEL_TYPE_CMD_TIME`. Even if it has a time it will be renewed.

The messages are send in [cs_MeshModel.cpp](https://github.com/crownstone/bluenet/blob/master/source/src/mesh/cs_MeshModel.cpp) where you have a separate function for sending messages, in this case `sendTime` and one that receives messages, `handleMsg`. Sending is done by adding the message to be send to a queue: `addToQueue`. The receiving messages are captured by a large switch statement, e.g. selecting `CS_MESH_MODEL_TYPE_STATE_TIME`. It's important to distinguish if its an own message or not (which is done through checking on `NRF_MESH_RX_SOURCE_LOOPBACK`).

All events are subsequently send through the system as an `event_t` through the `EventDispatcher`.

**Propagate user locations**
User locations can be propagated in a similar fashion. A user is represented by a "profile id" and a "location id". 

* Add the types to [cs_MeshModelPackets.h](https://github.com/crownstone/bluenet/blob/master/source/include/protocol/mesh/cs_MeshModelPackets.h)
* Add an additional event that represents profile/location tuples.
* Send the event through the event dispatcher from the [cs_BackgroundAdvHandler.cpp](https://github.com/crownstone/bluenet/blob/master/source/src/processing/cs_BackgroundAdvHandler.cpp).
* Add a sendPosition method to [cs_MeshModel.cpp](https://github.com/crownstone/bluenet/blob/master/source/src/mesh/cs_MeshModel.cpp).
* Call sendPosition when receiving corresponding event from event dispatcher.
* Add a switch case for receiving these messages in the same file. Send event through the event dispatcher as well.
* Add event to [PresenceHandler::handleEvent](https://github.com/crownstone/bluenet/blob/master/source/src/presence/cs_PresenceHandler.cpp).

vliedel, 2019-12-06:
So this implementation will not do any synchronization, right?

mrquincle, 2019-12-06:
This is not a clear question. :-) Interpretation. If the events are coming in at the cs_BackgroundAdvHandler they are forwarded into the mesh. Do we have any filtering if we do it like this? No.

Hence, your suggestion would be to:

- Generate the event not from the cs_BackgroundAdvHandler, but from the PresenceHandler. 
- Have the presence handler do filtering so not every broadcast message about presence will end up in the mesh, saturating it.

The filter can be something like:

- Send every X messages from the phone.
- Send only messages when something changed.

mrquincle, 2019-12-09:
The current event handler does not yet filter. This will be implemented by @ArrowAcrobatics.

mrquincle, 2019-12-09:
We will test this in the following way:

+ Create a compilation option to remove the (active) scanner functionality from the mesh.
+ We have a version of the consumer application that send broadcasts with profile and location id.

Now, we program two dev. boards, the first one receives the broadcasts from the phone and transmits its over the mesh, the second one receives them over the mesh. Just a print statement that indicates that the message has been received means that the test has been succesful.

closed 2019-12-09 by mrquincle