crownstone/bluenet #94: Cached cmake variables.

vliedel, 2020-10-06:
Some cmake variables are cached, while they shouldn't. The two most important ones are `CMAKE_BUILD_TYPE` and `FACTORY_IMAGE`.
We don't want people to make release builds unless they specify it, else non-release firmware might be on crownstones without being able to know.

mrquincle, 2020-10-06:
Those are two different things.

`CMAKE_BUILD_TYPE` is something internal to `cmake`. You don't have full control over it. The `CMAKE_BUILD_TYPE` will always be generated. The default is an empty string.

I've added an if-statement in the arm cmake file to make sure it is set to something (for the duration of compilation) by default  https://github.com/crownstone/bluenet/commit/f5740c8254efedfae01474550c79112364c476fd 

You might have a `CMAKE_BUILD_TYPE` variable set in your case. You will have to remove this once. For example through `cmake-gui`.

mrquincle, 2020-10-06:
The `FACTORY_IMAGE` is completely under our own control. I have not found `FACTORY_IMAGE` in my cache. I also don't see it to be `SET` anywhere. Are you sure your statement is correct?

vliedel, 2020-10-06:
No I have not verified the factory image, that's been ages ago, and i just
add `-DFACTORY_IMAGE=` and `-DCMAKE_BUILD_TYPE=Debug` every time i compile
anything.

mrquincle, 2021-01-27:
This issue seems to be a non-issue then, can it be closed?

vliedel, 2021-02-03:
Well, it's not being tested, i guess you can close it until it occurs again.

mrquincle, 2021-02-03:
Closed because of misunderstandings about the workings of `CMAKE_BUILD_TYPE`.

closed 2021-02-03 by mrquincle