crownstone/bluenet #95: Calling make depend in build directory doesn't recurse through all configs

ArrowAcrobatics, 2020-10-16:
Sometimes after changing brances, running `$GITDIR/bluenet/build make` fails with output such as
```
$GITDIR/bluenet/source/include/structs/buffer/cs_InterleavedBuffer.h', needed by 'CMakeFiles/crownstone.dir/src/behaviour/cs_BehaviourHandler.cpp.o'.  Stop.
```

A `$GITDIR/bluenet/build make depend` does not solve the issue because it somehow skips some configurations/targets. E.g.:

`bluenet/build$ make depend VERBOSE=1`
bluenet -B$GITDIR/bluenet/build --check-build-system CMakeFiles/Makefile.cmake 1
Clearing dependencies in "CMakeFiles/nrfconnect_programmer.dir/depend.make".
Clearing dependencies in "CMakeFiles/nrfconnect_core.dir/depend.make".
Clearing dependencies in "CMakeFiles/write_bootloader.dir/depend.make".
Clearing dependencies in "CMakeFiles/write_application.dir/depend.make".
Clearing dependencies in "CMakeFiles/list_jlinks.dir/depend.make".
Clearing dependencies in "CMakeFiles/nordic_sdk.dir/depend.make".
Clearing dependencies in "CMakeFiles/write_board_version.dir/depend.make".
Clearing dependencies in "CMakeFiles/gcc_arm_none_eabi.dir/depend.make".
Clearing dependencies in "CMakeFiles/micro_eec.dir/depend.make".
Clearing dependencies in "CMakeFiles/nordic_mesh_sdk.dir/depend.make".
Clearing dependencies in "CMakeFiles/bluenet.dir/depend.make".
Clearing dependencies in "CMakeFiles/create_git_release.dir/depend.make".
Clearing dependencies in "CMakeFiles/write_softdevice.dir/depend.make".
Clearing dependencies in "CMakeFiles/write_mbr_param_address.dir/depend.make".
Clearing dependencies in "CMakeFiles/read_softdevice_version.dir/depend.make".
Clearing dependencies in "CMakeFiles/create_git_release_bootloader.dir/depend.make".
Clearing dependencies in "CMakeFiles/erase.dir/depend.make".
Clearing dependencies in "CMakeFiles/nrfconnect_core_setup.dir/depend.make".
Clearing dependencies in "CMakeFiles/dev6415.dir/depend.make".
Clearing dependencies in "CMakeFiles/reset.dir/depend.make".
Clearing dependencies in "CMakeFiles/debug_application.dir/depend.make".
```

The output above is not truncated. Note that there is only one line of the form:
`Clearing dependencies in "CMakeFiles/dev6415.dir/depend.make".`, while I have 5 devboard configs. 

Running `cd $GITDIR/bluenet/build/dev1657 && make depend && make` succeeds. But only fixes that specific config,
Running `cd $GITDIR/bluenet/build/dev1657 && make` after the above command still fails with the output:
```
make[2]: *** No rule to make target $GITDIR/bluenet/source/include/structs/buffer/cs_InterleavedBuffer.h', needed by 'CMakeFiles/crownstone.dir/src/behaviour/cs_BehaviourHandler.cpp.o'.  Stop.
make[1]: *** [CMakeFiles/Makefile2:640: CMakeFiles/crownstone.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
```

Expected behaviour:
`cd $GITDIR/bluenet/build && make depend && make` should run without errors.

Possible temporary solution:
Add a githook to local repository to run a script that calls make depend in all build config dirs.

mrquincle, 2020-10-16:
**Current setup**
It is the case that it is quite hard to have `cmake` build for different targets. Note that the **configuration** is namely different per target. For one target it can be that debug is enabled, while for another target this is not the case. This is the reason that you have to run `cmake` if you want to configure your system for a different target rather than only running `make`. 

This is why there is a `-DBOARD_TARGET` as explained in the [docs](https://github.com/crownstone/bluenet/blob/master/docs/INSTALL.md) and for some more background info [here](https://github.com/crownstone/bluenet/blob/master/docs/BUILD_SYSTEM.md). 

If you would want to change this, then you'll have to give as an option a **list of targets**. The `cmake` system should then be able to configure every target simultaneously. This is quite a rewrite.

**First step, use cmake how it should be used**
As a first step I would then recommend the following rewrite. Remove most configuration flags that are passed to the compiler and the linker. Start using the configuration method that's inherent to `cmake` instead. The `CONFIGURE_FILE` paradigm allows you to create header files with custom fields which will be placed in the `build` directory (and not committed to version control).

**One way to do this**
Then you will still need to run the compiler multiple times over the same source directory, but with different targets. One way to do this is by having more `ExternalProject`'s defined in the parent `CMakeLists.txt` file. I don't know if it's easy to have this dynamic depending on the number of boards you want to compile for.

**Rebuild release builds**
Note that you also have to discuss this with the team. For example, for in particular **release** builds, it might be undesirable if building another target would also rebuild this release target. There might also be other reasons why people want to have more fine-grained control over which targets will be rebuild after a general `make` command.

**Running only make**
Note also that if you just run `make` for the individual directories after changing the configuration (that potentially might have changed the code or how they should be compiled), this is not sufficient. In that case you will also need to make sure `cmake` did run for all those targets as well.

vliedel, 2020-10-16:
I'd expect running `make depend` in de _build_ dir to do the same as running `make depend` in the _build/<target>_ dir.

mrquincle, 2020-10-16:
**Deployment information**
After some back and forth. The actual problem is the ability to upload the same code to say around 5 Crownstones. The current deployment information (which serial port to use, etc.) is part of the configuration process. This should **not** be hardwired to a particular target. This can be solved by a particular file that defines the deployment. Here we can have a list of targets mapped to JLink ids and debug ports. This means you don't need to create multiple targets just for the same binary which is just loaded into a different board.

**Feature request**
After that it would be useful to:

* use a CLI utility (e.g. the python tools) to set a Crownstone to DFU.
* use the Nordic GUI as explained [here](https://github.com/crownstone/bluenet/blob/master/docs/BUILD_SYSTEM.md) to upload the new firmware.

This would allow a developer to update firmware without using a dev. app or an RC.

mrquincle, 2020-10-16:
> I'd expect running `make depend` in de _build_ dir to do the same as running `make depend` in the _build/target_ dir.

Here `target` is defined by the parent `cmake` process. To run `make anything` only works in `build` and in `build/target`. It will be the case that `cmake` and `make` is completely oblivious to any previously configured target `build/previoustarget`. To run `make anything` in such an old target directory results in undefined behavior. Here `make` will not even know if files have changed, nor if configuration options have been changed. When switching from target you will **always** have to run `cmake` itself with the `-BOARD_TARGET=XXX` option.

mrquincle, 2021-01-27:
Making use of `configure_file` rather than passing everything as compiler flags is done by https://github.com/crownstone/bluenet/commit/71859ac33231f5c3fc4aed36f80aa47a230f5154, https://github.com/crownstone/bluenet/commit/e05d811448c1437837b5d789fc0585561a462b87, https://github.com/crownstone/bluenet/commit/e29933758102f23fd10382ff32600f179c313003, and is not finished yet.

Rewriting the build tool such that `GDB_PORT` is not defined at configuration time is done in https://github.com/crownstone/bluenet/commit/081a098b70406f704b07d50eaa284afe572e8b09 and is not finished yet either.

Both are trivial to complete and will solve the discussed issues (apart from the misunderstandings about what cmake actually does).

mrquincle, 2021-03-09:
Issue https://github.com/crownstone/bluenet/issues/111 describes this slightly differently. 

We can not iterate through all targets. The main `CMakeLists.txt` file is only configured for a single target. However, `make depend` should actually go into that particular target directory, which it doesn't.

Hence, now you'll need to run `make depend` in that target directory as well. It might be worth the time to quickly look into a possibility to run this command manually if it is hard to make it work recursively.

mrquincle, 2021-03-09:
Fixed in https://github.com/crownstone/bluenet/commit/56bf29e434c70d1688d9c556f2e48e1800834234

closed 2021-03-09 by mrquincle