crownstone/bluenet #96: Bug in _log

mrquincle, 2021-02-01:
The macro `_log` does not respect the `newline` argument anymore:

```#define _log(level, addNewLine, fmt, ...) \```

See https://github.com/crownstone/bluenet/blob/master/source/include/logging/cs_Logger.h#L229

This breaks `printAddress` in `csUtils.h` and results in the MAC address printed across several lines.

vliedel, 2021-02-05:
Combining that with the prefix (file + line nr) is not worth the effort for plain text logging, which is only there as fallback option.
We could remove the prefix, but I think that is valuable information when debugging.

mrquincle, 2021-02-05:
Now it is:

```
[t/source/src/cs_Crownstone.cpp : 705 ] Address: 
[source/include/util/cs_Utils.h : 81  ] C2:
[source/include/util/cs_Utils.h : 81  ] 81:
[source/include/util/cs_Utils.h : 81  ] 75:
[source/include/util/cs_Utils.h : 81  ] 46:
[source/include/util/cs_Utils.h : 81  ] D8:
[source/include/util/cs_Utils.h : 83  ] F1
[t/source/src/cs_Crownstone.cpp : 707 ] Address id=0 type=1

Address: C2:81:75:46:D8:F1
```

I don't think you grokked the bug. You print a new line on every line. The `addNewLine` parameter is ignored.

mrquincle, 2021-02-05:
I would expect something

```
#define _log(level, addNewLine, fmt, ...) \
				if (level <= SERIAL_VERBOSITY) { \
                                        // some if statement on addNewLine
					cs_log_printf("[%-30.30s : %-4d] " fmt "\r\n", _FILE, __LINE__, ##__VA_ARGS__); \
				}
```

vliedel, 2021-02-05:
Yes, I know what you mean, but without newline, you will get:

[t/source/src/cs_Crownstone.cpp : 705 ] Address:
[source/include/util/cs_Utils.h : 81  ]
C2:[source/include/util/cs_Utils.h : 81  ]
81:[source/include/util/cs_Utils.h : 81  ]
75:[source/include/util/cs_Utils.h : 81  ]
46:[source/include/util/cs_Utils.h : 81  ]
D8:[source/include/util/cs_Utils.h : 83  ] F1

mrquincle, 2021-02-05:
Fixed by adding `prefix` argument to `#define _log(level, prefix, addNewLine, fmt, ...)`

closed 2021-02-05 by mrquincle

mrquincle, 2021-02-05:
Hereby I declare commit https://github.com/crownstone/bluenet/commit/42263bfc5352b19de82ba18c272c7de3f586175e under protest. 

I advice against writing to a `prefix` static boolean depending on  `addNewLine` being true.

```
#define _log(level, addNewLine, fmt, ...)
```

Now to write a banner or something else you can't use write

```
_log(INFO, false, true, "text")
_log(INFO, false, true, "text")
_log(INFO, false, true, "text")
```

This is because it will start to introduce a `prefix` because you set the `addNewLine` parameter. 

It's a magic parameter you can manipulate through writing the new line yourself.

```
_log(INFO, false, false, "text\n")
_log(INFO, false, false, "text\n")
_log(INFO, false, false, "text\n")
```
Now, the suffix will not be added...

AlexDM0, 2021-02-05:
How about a _log, _logLn and _logBanner define to get rid of those ugly booleans in between?

mrquincle, 2021-02-05:
You will need `_log`, `_logLn`, `_logPrefix`, `_logPrefixLn`, but sure. That's also explicit. I prefer anything above magic.