crownstone/bluenet #97: Does not advertise when BUILD_MESHING=0

mrquincle, 2021-02-01:
There's something wrong with advertising when meshing is disabled.

The `init` function in cs_Advertiser.cpp is called. It must be something else.

Minor: update terminology in following piece of code. Now it seems like this is an error condition if verbosity of the class is increased.

```                                                      
   if (_advHandle != BLE_GAP_ADV_SET_HANDLE_NOT_SET) {                                                                 
      LOGAdvertiserVerbose("sd_ble_gap_adv_set_configure without params");                                            
      err = sd_ble_gap_adv_set_configure(&_advHandle, &_advData, NULL);                                               
      APP_ERROR_CHECK(err);                                                                                           
   }
``` 

Adjust to something like `LOGAdvertiserVerbose("Configure new advertisement (cycles between different ones)");`

vliedel, 2021-02-03:
IBeacon advertising is done via the mesh. Are you sure you don't see the
service data advertisements with BUILD_MESHING=0 ?

mrquincle, 2021-02-03:
It looks like it. 

It might be the case that I have removed a define that is used somewhere in the code. I've checked everything a few times though. 

If I'm correct there was no other flag that explicitly disabled the service data advertisements. There was only `IBEACON` and `CONFIG_IBEACON_ENABLED_DEFAULT`. I also didn't forget `SCAN_INTERVAL` and `SCAN_WINDOW`. I even checked `BOOT_DELAY` just in case it had to do with some kind of racing condition w.r.t. enabling the radio.

Hence, I know you've often enabled the mesh and you were busy with setting up connections, so that's why my guess was that it was broken before I started. :-)

vliedel, 2021-02-19:
I can't reproduce this. With `BUILD_MESHING=0` and `MESHING=0`, i still see advertisements with service data, both in setup and normal mode.

mrquincle, 2021-02-19:
I can't reproduce this anymore either, will close for now.

closed 2021-02-19 by mrquincle