crownstone/bluenet #98: Writing to serial should be limited

mrquincle, 2021-02-04:
A call to `cs_log_printf` is not restricted in size.

```
#1  0x000301e6 in cs_log_printf (str=0x60e66 "[%-30.30s : %-4d] heapEnd=0x%X maxHeapEnd=0x%X minStackEnd=0x%X minFree=%u sbrkFails=%u\r\n") at /home/anne/workspace/bluenet/source/src/logging/cs_Logger.cpp:53
```

It should make use of the outgoing uart buffer. The C and C++ code should not use different buffers. There should not be dynamic memory allocation for print statements.

mrquincle, 2021-02-04:
This particular call allocates 520 bytes and frees 520 bytes in quick succession.

mrquincle, 2021-02-04:
Another reason not do memory allocation in print statements, is that it makes it harder to put print statements in memory allocation routines.

vliedel, 2021-02-05:
cs_log_printf is meant as a convenience in case you don't want to use
binary logging. If you care about memory and speed, use binary logging
instead.
An option would be to simply ignore any cs_log_printf that would need more
than 128B, or maybe we could use cs_clog instead, which streams directly to
uart instead of to a buffer first.

mrquincle, 2021-02-05:
Check. The buffer of size `UART_TX_BUFFER_SIZE` has a different function that I anticipated. It is used as an intermediate buffer to write into from, for example, the ADC. This can be done **asynchronously**. That is, in the meantime other things can be written to the console. Hence, this buffer is hard to re-use for this purpose.

Easy to implement would then be the removal of `malloc` for **non binary** logging. Hence, just use `snprintf` and write up to `n`. Due to the fact that `__func__` takes many bytes, it might be worthwhile to only print the data without prefix.

vliedel, 2021-02-05:
Fixed with 45b984ce3b2f005cbd7fa3a674722cc1b902e00d

closed 2021-02-05 by vliedel