crownstone/bluenet #106: Feature request: remove virtual pins

mrquincle, 2021-02-19:
The current code is motivated by a standard blinking example. I thought it would be great to have https://www.arduino.cc/en/Tutorial/BuiltInExamples/Blink working on the Crownstone but then with your normal lights. Just replace `LED_BUILTIN` with `CROWNSTONE_RELAY`.

```
// the setup function runs once when you press reset or power the board
void setup() {
  // initialize digital pin CROWNSTONE_RELAY as an output.
  pinMode(CROWNSTONE_RELAY, OUTPUT);
}

// the loop function runs over and over again forever
void loop() {
  digitalWrite(CROWNSTONE_RELAY, HIGH);   // turn the light on
  delay(1000);                            // wait for a second
  digitalWrite(CROWNSTONE_RELAY, LOW);    // turn the light off
  delay(1000);                            // wait for a second
}
```

This feature request by @vliedel and @AlexDM0 asks for removal of this functionality. Hence, do not use it. 

The syntax will be more something like (this is custom and can change depending on lib version):

```
#include <Crownstone.h>

setup() {
   // some initialization
   Crownstone.init();
}

loop() {
    Crownstone.switch(1);
    delay(1000);
    Crownstone.switch(0);
    delay(1000);
}
```

There are a few reasons for this:

* You will be confused in thinking that `CROWNSTONE_RELAY` is an actual GPIO pin on the board.
* The Crownstone switch can be **locked**. In that case writing something over `digitalWrite` won't have an effect. You will need to call a function like `Crownstone.isLocked()` and `Crownstone.unlock()` before `Crownstone.switch(1)`.
* The Crownstone dimmer can be **disabled**. Using an `analogWrite` would then have no effect.

There might be a middle way in providing both.

* I think to have those examples working as is would be sweet. 
* Those examples are the ones that you recognize as Arduino developer. 
* More dedicated functions (to capture corner case, see above) can always be provided through a Crownstone lib.

Just don't get your hopes up. :-D

AlexDM0, 2021-02-19:
If an external sensor triggers an external activator or light or led via the pins on the header you can do the blinky exactly as you have written it:
```
// the setup function runs once when you press reset or power the board
void setup() {
  // initialize digital pin HEADER_LED as an output.
  pinMode(HEADER_LED, OUTPUT);
}

// the loop function runs over and over again forever
void loop() {
  digitalWrite(HEADER_LED, HIGH);   // turn the light on
  delay(1000);                            // wait for a second
  digitalWrite(HEADER_LED, LOW);    // turn the light off
  delay(1000);                            // wait for a second
}
```

mrquincle, 2021-02-19:
Most Arduino users will not hook up the Crownstone board physically with other things. 

You are describing a scenario where people add sensors to a 230V board through soldering a connector on the place of the GPIO pins and then connecting leds to those. 

I am talking about the on-board **switch** and **dimmer**.

closed 2021-11-09 by mrquincle