crownstone/bluenet #124: Feature request: Do not hardcode debug verbosity level in .cpp or .h files themselves

mrquincle, 2021-09-09:
**Observation**
Currently there are many statements like this, for example, in `cs_SmartSwitch.cpp`

```
#define LOGSmartSwitchDebug LOGvv
```

They have to be changed when working on the code and then changed back when you commit. 

**Solution direction**
This can be easily added to the build system itself. Add to a file like `cs_StaticConfig.h.in` the following:

```
#define LOGSmartSwitchDebug @LOG_SMART_SWITCH_DEBUG@
```

And then define in `CMakeBuild.config.default` the default debug settings.

```
# Debug settings per module
LOG_SMART_SWITCH_DEBUG=LOGvv
```

Overwrite them in your local config file and voila! 

**Other reasons**
If we decouple it like this, we have some more flexibility. 

+ We can have different verbosity levels depending on the target.
+ We can have different defaults depending on the hardware (e.g. nRF52832 vs nRF52840). 
+ We can have different defaults depending on the type of product.
+ We can easily share a config file with someone else to have exactly the same configuration without having to commit hardcoded verbosity levels in the code.

That apart from aesthetic reasons. :-)

vliedel, 2021-09-13:
Sounds like a good idea to me.

Maybe it's better to set verbosity level in the config file though (`LOG_SMART_SWITCH_DEBUG=SERIAL_VERY_VERBOSE`), so we can also use that level for `_log()` and `_logArray()`. The define of `LOGSmartSwitchDebug` can then be done in  *cs_StaticConfig.h* like `#define LOGSmartSwitchDebug(fmt, ...) _log(LOG_SMART_SWITCH_DEBUG, true, fmt, ##__VA_ARGS__)`.

However, this has the downside that adding a log definition takes more time, and that `NRF_LOG_DEBUG` is always used for these logs.

mrquincle, 2021-09-13:
Let's summarize:

- A lot of the complexity comes from trying to have macros resolve to functions or **empty macros** to make sure the more verbose options are not compiled at all. 
- Most important is not to have unnecessary log/debug strings in a binary. Especially for our use case, this would mean large firmware blobs.
- We use macros so we have access to `__LINE__`, etc. at the right place (there where the print statement is defined, not in the logger header file).
- The linker (given the right settings) throws out the functions that aren't used, however, the above means we can't do it in the form of functions.
- We want global verbosity levels as well as verbosity levels per module. The verbosity level per module is overruled by the global verbosity level. This means that e.g. in release we don't have suddenly some module that is still able to write logs. 

If there is some way in which we can write `Log(SmartSwitchDebug, "set %u", intensity);` rather than `LOGSmartSwitchDebug("set %u", intensity);` while keeping the above in mind, it would be awesome. 

I think this will address your concern that otherwise it might take quite some time to add a logging definition.

mrquincle, 2022-08-29:
Will solve this issue after microapp.

It will change things like:

```
	if (scannedDevice->rssi > RSSI_LOG_THRESHOLD) {
#ifdef COMMAND_ADV_VERBOSE
		LOGCommandAdvVerbose("rssi=%i", scannedDevice->rssi);
		_log(SERIAL_DEBUG, false, "16bit services: ");
		_logArray(SERIAL_DEBUG, true, services16bit.data, services16bit.len);
#endif
#ifdef COMMAND_ADV_VERBOSE
		_log(SERIAL_DEBUG, false, "128bit services: ");
		_logArray(
				SERIAL_DEBUG,
				true,
				services128bit.data,
				services128bit.len);  // Received as uint128, so bytes are reversed.
#endif
	}
```

Into:

```
	if (scannedDevice->rssi > RSSI_LOG_THRESHOLD) {
		LOGv(COMMAND_ADV_HANDLER, "rssi=%i", scannedDevice->rssi);
		_log(COMMAND_ADV_HANDLER, SERIAL_DEBUG, false, "16bit services: ");
		_logArray(COMMAND_ADV_HANDLER, SERIAL_DEBUG, true, services16bit.data, services16bit.len);

		_log(COMMAND_ADV_HANDLER, SERIAL_DEBUG, false, "128bit services: ");
		_logArray(
				COMMAND_ADV_HANDLER,
				SERIAL_DEBUG,
				true,
				services128bit.data,
				services128bit.len);  // Received as uint128, so bytes are reversed.
	}
```

The code that defines the verbosity level for the `COMMAND_ADV_HANDLER` can be through macros. As long as it doesn't introduce conditional precompiler statements, it's fine.

vliedel, 2022-08-29:
Can we first check whether most consider this an issue? I'm ok with how it's done right now, and i'm not so sure the proposal is going to be an improvement for the developer. It has the downsides that adding a module log level becomes a bit harder, and changing the log level as well (can't just click on the log level anymore).

mrquincle, 2022-08-29:
I agree that it shouldn't be made harder.

mrquincle, 2022-08-29:
Currently https://github.com/crownstone/bluenet/blob/master/source/src/microapp/cs_MicroappStorage.cpp is a better example of how the code should look like. There are no conditional statements in there.

Just a matter of defining:

```
#define LOGMicroappInfo LOGi
#define LOGMicroappDebug LOGvv
#define LOGMicroappVerboseLevel SERIAL_VERY_VERBOSE
```

And used like this:

```
	LOGMicroappDebug("binStartAddress=0x%08X", binStartAddress);
```

There are some pros and cons.

* In my opinion verbosity levels shouldn't be hardcoded in the code itself. A more proper place would be some configuration file. This is also how Nordic does it. There's an `sdk_config.h` and `app_config.h` file in which the default log levels are written down for the SDK and the overwrites are written down for the developed firmware. 
* The use of `cmake` to configure verbosity levels would make it impossible for @vliedel to click on a verbosity level to change it.
* The use of a header file like `app_config.h` for verbosity levels would for any change require recompilation of all compilation units that include this file.
* If the log level changes for a module with the current setup, only recompilation of that module will be triggered.
* If we would use `cmake` we can have the debug levels in the config files for particular targets. It is easy to commit the defaults and overwrite those defaults with local configurations.

A possibility is to allow for a C header file which is kept local. Everything that is just required for local development can be adjusted and doesn't run the risk of being committed. 

We can do this even **per module** through the C++17 `__has_include` feature. Fancy!

mrquincle, 2022-08-30:
In https://github.com/crownstone/bluenet/tree/maintenance/logging a proposal has been made with the `__has_include` feature.

```
// This sets logging macros 
#include <util/cs_ResetMacros.h>

// This can overwrite things locally. Create a XXX.local.h file to do so.
#if defined __has_include && __has_include(<uart/cs_UartCommandHandler.local.h>)
#include <uart/cs_UartCommandHandler.local.h>
#endif

// In the code now things like these can be used
LOGDebug("Handle cmd opCode=%u size=%u accessLevel=%u", opCode, commandData.len, accessLevel);
```

If the verbosity level has to be adjusted, this can be done in the header file:

```
#undef LOGInfo
#define LOGInfo LOGi
```

However, it's not according to the requirements by @vliedel who wants to click on a verbosity level to change it and not create a header file and write down the temporarily verbosity level.

I assume this means this will never be fixed and this type of information will remain hardcoded. 

- I don't know how to convey that using `#define LOGInfo LOGi` rather than `#define LOGMicroappInfo LOGi` is in general to be preferred. Given we are already in a microapp file, the repetition of the term Microapp in macro names seems unnecessary. Some arguments. The code gets shorter. Copying a log statement doesn't need rewrites depending on the context. It's similar to using `buffer->read` rather than `buffer->readBuffer`.
- I don't know how to convey that having developer toggles such as debug levels like this in the code base itself is something to prevent.

I will close this issue because I don't see a way forward. Feel free to reopen if there's a constructive way to implement this.

closed 2022-08-30 by mrquincle

mrquincle, 2022-08-30:
Also, one last note. The verbosity levels are not only defined on the level of a compilation unit, but also across compilation units. See e.g. `cs_MeshDefines.h`:

```
// Debug logs
#define LOGMeshWarning LOGw
#define LOGMeshInfo LOGvv
#define LOGMeshDebug LOGvv
#define LOGMeshVerbose LOGvv
#define LogLevelMeshDebug SERIAL_VERY_VERBOSE

// Debug logs
#define LOGMeshModelInfo LOGvv
#define LOGMeshModelDebug LOGvv
#define LOGMeshModelVerbose LOGvv
#define LogLevelMeshModelVerbose SERIAL_VERY_VERBOSE
```

This means that `LOGInfo` as an adjustable redirection for `LOGi` won't work because there are many possible `logging channels` per compilation unit.