crownstone/bluenet #129: Bug in uart binary client

mrquincle, 2021-10-05:
**Describe the bug**

On running  `make uart_binary_client` the build system defaults to errors.

```
UART client, runs in background
** UART client, keep running in separate shell
Listening for logs on port /dev/ttyACM0, and using files in "/home/anne/workspace/bluenet/source" to find the log formats.
Traceback (most recent call last):
  File "/home/anne/workspace/bluenet/scripts/log-client.py", line 44, in <module>
    bluenetLogs.setSourceFilesDir(sourceFilesDir)
  File "/home/anne/.local/lib/python3.8/site-packages/bluenet_logs/BluenetLogs.py", line 20, in setSourceFilesDir
    self.logLineRetriever.setSourceFilesDir(dir)
  File "/home/anne/.local/lib/python3.8/site-packages/bluenet_logs/LogLineRetriever.py", line 39, in setSourceFilesDir
    self._cacheFileContents()
  File "/home/anne/.local/lib/python3.8/site-packages/bluenet_logs/LogLineRetriever.py", line 82, in _cacheFileContents
    lines = file.readlines()
  File "/usr/lib/python3.8/codecs.py", line 322, in decode
    (result, consumed) = self._buffer_decode(data, self.errors, final)
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xd3 in position 7734: invalid continuation byte
make[3]: *** [CMakeFiles/uart_binary_client.dir/build.make:59: CMakeFiles/uart_binary_client] Error 123
make[2]: *** [CMakeFiles/Makefile2:272: CMakeFiles/uart_binary_client.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:279: CMakeFiles/uart_binary_client.dir/rule] Error 2
make: *** [Makefile:229: uart_binary_client] Error 2
```

**To Reproduce**

Checkout and run  `make uart_binary_client`.

**Expected behavior**

Working binary UART client.

**Already done**

At least on an error like this the user should be hinted in setting the `-DDOWNLOAD_BLUENET_LIB_LOGS=ON` option for cmake. 

Not that that fixed it. :-)

mrquincle, 2021-10-05:
It looks like every single file in the working directory is opened.

```
	def _cacheFileContents(self):
		"""
		Cache contents of all cached fileNames.
		"""
		for fileName in self.fileNames:
			filePath = fileName
			try:
				file = open(filePath, "r", encoding="utf-8")
				lines = file.readlines()
				file.close()
			except:
				_LOGGER.warning(f"Error in file {filePath}")
			self.bluenetFiles[fileName] = lines
```

Only source files should be inspected. 

It currently fails on a `tags` file which is generated by `ctags`. However, that's quite irrelevant. There can be other files here as well that are not source files.

mrquincle, 2021-10-05:
Options to fix:

# Walk the source and include directories

Walk it manually and pick up `.c`, `.cpp` and `.h` files and a few other extensions that might be useful.

# Cache files in cmake

Make use of the cache files that are populated by `cmake`.

```
cat CMakeFiles/crownstone.dir/CXX.includecache | sort | uniq | grep -v '#'
```
This lists:

```
access_config.h
access.h
access_internal.h
access_publish.h
access_reliable.h
access_status.h
...
...
...
util/cs_Utils.h
util/cs_UuidParser.h
util/cs_Variance.h
util/cs_WireFormat.h
utility
utils.h
vector
```
Here you can use a regexp if you'd like to excluded Nordic files or other files.

# Variables in cmake

Alternatively, you can use the CMake variable we have defined to collect all source files:

```
cat list_variables.txt | grep FOLDER_SOURCE
```
This does not include header files, so it's less useful.

mrquincle, 2021-11-22:
@vliedel This is still not fixed. If there is any binary (or even non-binary) file in the directories, the binary logger can choke on it.

closed 2021-11-26 by vliedel

vliedel, 2021-11-29:
Wasn't that fixed by bluenet-logs 0.1.4 ?