crownstone/bluenet #142: Example code for tracking wearables

hillstub, 2021-11-23:
Hi all,
I just went through the process of upgrading my crownstones, hoping to enable tracking my wearables. 
However, it's a bit unclear to my how to achieve this.
Could you post some example code?
Thanks in advance!

vliedel, 2021-11-24:
Hi, the asset filters are indeed not aimed for home usage yet: there is no integration in the app or in the hub yet. Next to that, the crownstones themselves can only determine a wearable being in the sphere, not which room.

With that all said, there is a [python library example](https://github.com/crownstone/crownstone-lib-python-uart/blob/master/examples/asset_filter_example.py) to configure an asset filter. In some cases, this will require you to know a bit about Bluetooth LE.

hillstub, 2021-11-24:
Thanks @vliedel !

I thought it would be possible to see the closest crownstone to the wearable - is that not the case?

I'm happy to dive into some python scripts, but could you elaborate a bit on the example? If I understand this correctly, this would upload a filter to the crownstone mesh for detecting wearables. Do I need another script to receive messages in case of an event happening?
Thanks!

vliedel, 2021-11-24:
When you look in the AssetFilter class, you will find an experimental optimization strategy "nearest". This means the crownstones will only send a mesh message when they *think* to be the nearest. Since this is not always correct (nearest can change over time, etc), we didn't commit on this strategy yet. With the current firmware, the best option is to gather all info (so without the optimization strategy) at a hub, and determine which crownstone is closest on the hub.

There is no example yet for receiving the asset filter reports, basically you will need to subscribe for `UartTopics.assetIdReport` (or `UartTopics.assetTrackingReport`, but i see now that this name should be changed to `assetMacReport`). Quick example:
```python
		def onAssetMac(report: AssetMacReport):
			print(f"onAssetMac mac={report.assetMacAddress} crownstoneId={report.crownstoneId} rssi={report.rssi} channel={report.channel}")

		def onAssetId(report: AssetIdReport):
			print(f"onAssetId id={report.assetId} crownstoneId={report.crownstoneId} rssi={report.rssi} channel={report.channel} passedFilterIds={report.passedFilterIds}")

		UartEventBus.subscribe(UartTopics.assetTrackingReport, onAssetMac)
		UartEventBus.subscribe(UartTopics.assetIdReport, onAssetId)
```

hillstub, 2021-11-24:
Thanks @vliedel!

I tried your code, but it report on my devices.
When enabling debugging I see
```
DEBUG  : Unknown opCode: 10111
```
which seems to refer to NEIGHBOUR_RSSI.
I do have the latest version of crownstone-uart (2.2.0)

vliedel, 2021-11-25:
Then nothing passes the filters you set, or the filters failed to upload. The unknown opcode is unrelated.
Did you change the example filters to match your use case?

hillstub, 2021-11-25:
Hi @vliedel,
I actually got it working with a Fitbit! I have two additional questions:
- Is there a way to filter on UUID instead of MAC address?
- Is there a way to throttle the filter? (e.g. only relay messages across the mesh every x seconds)

vliedel, 2021-11-26:
Nice :)

1. Depends on what UUID you mean, but you can filter on anything that is in the BLE advertisement. The AssetFilter class has some methods for common use cases, but `filterByAdData()` is the most raw one. [More info about BLE advertisement data](https://github.com/crownstone/bluenet/blob/master/docs/ASSET_FILTERING.md#ble-advertisement). I usually use the [nRF Connect app](https://www.nordicsemi.com/Products/Development-tools/nrf-connect-for-mobile) to view the raw advertisements of devices.
2. Not at the moment. The idea is that the firmware will automatically throttle when needed, but that is not implemented yet. Currently each asset is always throttled by 1 per second.

mrquincle, 2021-12-01:
@hillstub has this been answered according to your expectations? As soon as we formally release this functionality we will create user-facing examples. Feel free to reopen then.

closed 2022-08-12 by mrquincle