crownstone/bluenet #151: Enforce modern version of cmake

mrquincle, 2022-01-31:
The `cmake` build tool is versatile, but the syntax is ugly. We're supporting quite an old `cmake` version. Many files from Zephyr (which we will be relevant for supporting Matter on the nRF52840) state:

```
cmake_minimum_required(VERSION 3.20.0)
```

Hence, we shouldn't stick to the cmake version supplied with an operating system. For example, if you're on LTS for Ubuntu, you'll be at 20.04 which has cmake version 3.16.3: https://launchpad.net/ubuntu/focal/+source/cmake. You can however easily update to a more modern version of cmake without having to compile anything yourselves. Check the instructions yourself on https://apt.kitware.com/.

```
sudo apt remove cmake
wget -O - https://apt.kitware.com/keys/kitware-archive-latest.asc 2>/dev/null | gpg --dearmor - | sudo tee /usr/share/keyrings/kitware-archive-keyring.gpg >/dev/null
echo 'deb [signed-by=/usr/share/keyrings/kitware-archive-keyring.gpg] https://apt.kitware.com/ubuntu/ focal main' | sudo tee /etc/apt/sources.list.d/kitware.list >/dev/null
sudo apt update
sudo rm /usr/share/keyrings/kitware-archive-keyring.gpg
sudo apt install kitware-archive-keyring
sudo apt install cmake
```

If we bump the `cmake` version we can for example remove a lot of the calculations around hexadecimal values (which is a moderately recent addition to `cmake`). 

If we do so, we of course have to instruct people on how to get a more modern version of `cmake` on their system. When we've described this properly it's time to enforce it.

mrquincle, 2022-02-28:
This can remove all from_hex and to_hex helper functions.

vliedel, 2022-03-01:
An Ubuntu LTS is supported for 5 years, so 18.04 is still a valid version.
However, it's quite simple indeed to upgrade the cmake version, so i'm ok with bumping the requirement to 3.20.

mrquincle, 2022-03-16:
This is part of branch https://github.com/crownstone/bluenet/tree/cmake-and-runtime but PR will only be opened after acceptance of current open ones.

closed 2022-04-12 by mrquincle