crownstone/bluenet #154: RTT logging in bootloader does not work well.

vliedel, 2022-02-14:
With the following config:
```
CS_SERIAL_ENABLED=SERIAL_ENABLE_RX_AND_TX
SERIAL_VERBOSITY=SERIAL_INFO
CS_UART_BINARY_PROTOCOL_ENABLED=1
CS_SERIAL_NRF_LOG_ENABLED=0
CS_SERIAL_NRF_LOG_PIN_TX=6
CS_SERIAL_BOOTLOADER_NRF_LOG_ENABLED=1
```

I do not always get RTT logs from the bootloader. I only seem to get bootloader RTT logs when:
- No application has been uploaded.
- Debugging the bootloader (`make debug_bootloader`).

Other things I tried:
- RTT logs from the application work fine (`CS_SERIAL_NRF_LOG_ENABLED=1`).
- UART logs from the bootloader work fine (`CS_SERIAL_BOOTLOADER_NRF_LOG_ENABLED=2`).
- Adding `NRF_LOG_FINAL_FLUSH();` just before `app_start();` in nrf_bootloader_app_start_final.c did not help.

mrquincle, 2022-02-14:
Perhaps https://devzone.nordicsemi.com/f/nordic-q-a/20708/rtt-logging-from-application-started-by-bootloader helps. The JLink page provides quite some information on RTT: https://www.segger.com/products/debug-probes/j-link/technology/about-real-time-transfer/ but it narrows down to setting the `.rtt_cb` and `.rtt_buffers` sections. 

Plain UART is almost always easier. As is just using `gdb` to step through stuff.

mrquincle, 2022-09-20:
There are issues with debugging that have to do with bugs in the Nordic code. Would say about this: won't fix.

closed 2022-09-20 by mrquincle